ros-visualization/rviz #1802: Fullscreen button for ImageDisplay

ClementLeBihan, 2023-09-15:
Hi ! 
I'm trying to add a fullscreen button for ImageDisplay, without success. 
I think it could be a good feature.

I've created a fullscreen button in the titlebar of the widget : 
```
QToolButton* fullscreen_button = new QToolButton();

  fullscreen_button->setText("FullScreen");
  fullscreen_button->setIconSize(QSize(10, 10));

  connect(fullscreen_button, SIGNAL(clicked()), this, SLOT(updateFullScreen()));

  dynamic_cast<QHBoxLayout*>(getAssociatedWidgetPanel()->titleBarWidget()->layout())
      ->insertWidget(2, fullscreen_button);
```

and implemented the updateFullScreen function : 
```
void ImageDisplay::updateFullScreen()
{
  if (!(getAssociatedWidgetPanel()->windowState() & Qt::WindowFullScreen)) {
    getAssociatedWidgetPanel()->setFloating(
        true);
    getAssociatedWidgetPanel()->showFullScreen();
  } else {
    getAssociatedWidgetPanel()->setFloating(false);
    getAssociatedWidgetPanel()->showNormal();
  }
}
```

Unfortunately, something weird happen. 
The first time I click, the widget start floating but not fullscreen. 
The second time I click, the widget doesn't move
The third time I click, the widget stop floating
The Fourth time I click, the widget goes fullscreen as expected. 

Do you have an idea why It needs 4 clicks ? 
I have no idea how QtWidget works, I may forgot to call a function ... 

Thank you for your help,
Best regards,
Clément

rhaschke, 2023-09-16:
I'm afraid that is a Qt limitation. See the corresponding note in the [docs](https://doc.qt.io/qt-5/qwidget.html#showFullScreen).
Maximizing the window works as expected: #1803
Note that you should remember the previous state of the display (floating or not) and restore that state.

ClementLeBihan, 2023-09-16:
In my side maximizing also need 4 clicks before beeing maximized. But there is the same warning about maximizing. Thank you for giving me the link, I wasn't aware of the limitation. 

It says `On X11, a window does not have a frame until the window manager decorates it. This happens asynchronously at some point in time after calling [QWidget::show](https://doc.qt.io/qt-5/qwidget.html#show)() and the first paint event the window receives, or it does not happen at all.` It seems to correspond the fact that I need to click 4 times on my button to see window maximizing or fullscreening. 

Thanks a lot for your quick answer and for your help @rhaschke !!!

rhaschke, 2023-09-17:
> On my side maximizing also needs 4 clicks before being maximized.

I was using Qt 5.15 on Ubuntu Jammy. Maybe on Focal with Qt 5.12 it doesn't (yet) work?

ClementLeBihan, 2023-09-18:
That's may be the point ! Thank you @rhaschke

closed 2023-09-18 by ClementLeBihan