ros-visualization/rviz #1805: Slow speed using ASan

captain-yoshi, 2023-09-21:
I am trying to pinpoint a bug from a _rviz_plugin_ that I created. I'm using the ASan sanitizer to try to locate it. But it is too slow to replicate the bug, because I need to save waypoints from an interactive marker. Built in Release mode with gcc and g++. Tested with a local build of RViz (d560273e15a518b31b30dcc4e6814511fd672a9b).

I currently add these flags `-fno-omit-frame-pointer -fsanitize=address` to the target_link_libraries of RViz and my plugin. I can see some outputs relatively fast when I click a QT button with a known bug. But using the interactive marker to move the robot is too slow.

Is there a way to only compile against ASan with my shared library (plugin) and not the rviz binary ? If I don't add the flags to RViz, I do not see any output from ASan. I know I can turn off some functions using special attributes, but I would prefer to not go this way. Obviously, I could store some waypoints in the code instead of relying of the interactive marker.

The binary is launched through roslaunch:
```cmake
 <node name="rviz" pkg="rviz" type="rviz" respawn="false" args="$(arg my_rviz)" output="screen" launch-prefix=""/>
```
I have also tried exporting before launching the RViz binary (without the flags) but the program crashes instantly (maybe because it does that for all the nodes...)
```sh
export LD_PRELOAD=/usr/lib/gcc/x86_64-linux-gnu/9/libasan.so
```

rhaschke, 2023-09-21:
I'm afraid, asan usage is slow due to the memory book-keeping overhead. You can't really avoid that. `LD_PRELOAD`, when applied to roslaunch, will sanitize all programs started - not a good idea.

captain-yoshi, 2023-09-21:
Thanks! Cleaned up my workspace for the 9th time and now it is much better, e.g. I can move the robot. So I guess I had one or more packages still in Debug...

closed 2023-09-21 by captain-yoshi