ros-visualization/rviz #1809: Wrench: Hide small values doesn't work as configured

ruoshiwen, 2023-11-14:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ruoshi-touchlab'
[ INFO] [1699978478.388180982]: rviz version 1.14.20
[ INFO] [1699978478.388205664]: compiled against Qt version 5.12.8
[ INFO] [1699978478.388212452]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1699978478.394047975]: Forcing OpenGl version 0.
[ INFO] [1699978478.563667668]: Stereo is NOT SUPPORTED
[ INFO] [1699978478.563713189]: OpenGL device: Mesa Intel(R) UHD Graphics (TGL GT1)
[ INFO] [1699978478.563726313]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
    ```
* If source build, which git commit? No
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: C.UTF-8
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!
  
  I am using rviz to display geometry_msgs/WrenchStamped message from F/T sensors. I would like to show small values so I unticked `Hide Small Values` and the display looked fine, so I save the config file as config.rviz. Here you can see a little red/yellow dot showed up, and that's how I think the `Hide Small Values` is disabled.
  
[image]

  
  However, whenever I open rviz again, I loaded the saved config file, although the box after Hide Small Values was kept unticked but no red/yellow dot showed up and the small values were hidden. So I ticked and unticked, and it worked. Here's a video showing this:

https://github.com/ros-visualization/rviz/assets/28282183/b9b2f30a-f798-44bc-a2a5-7c12f6f7847a

I also check the config file, 
`      Name: B12
      Queue Size: 10
      Topic: /thumb_00
      Torque Arrow Scale: 2
      Torque Color: 204; 204; 51
      Unreliable: false
      Value: true
    - Alpha: 1
      Arrow Width: 0.009999999776482582
      Class: rviz/WrenchStamped
      Enabled: true
      Force Arrow Scale: 0.00800000037997961
      Force Color: 204; 51; 51
      Hide Small Values: false
      History Length: 1`
     
The Hide Small Values is set to false. 
Could anyone help me here? I have tens of sensors to visualise at the same time, I don't want to tick and untick that box every time I start rviz.

closed 2023-11-15 by rhaschke

ruoshiwen, 2023-11-16:
Hi, thank you for the prompt action. I install rviz using 
`sudo apt-get install ros-noetic-rviz`
I run the rviz again with the config which set the "Hide Small values" to false, but it's still the same. Maybe I need to install rviz in a different way?
Look forward to your reply.

rhaschke, 2023-11-16:
These changes will be available from the binary package only after the next release, which I didn't even cut.
Meanwhile, I suggest building rviz from source. Just clone this repo into your ROS workspace.

PR #1810 (2023-11-15): Fix screw display
Fixes #1809