ros-visualization/rviz #1815: Failed to build rviz from source due to `SIP` version issue?

NumbNutN, 2024-02-02:
* OS Version: Ubuntu 20.04 LTS
* ROS Distro: Noetic
* RViz version: noetic-devel commit b466cb9d21a78170d9031cbbc7cc5ecc9886af52
* Qt version: 5.15.2

Hi, I was confused when I tried to build the RViz from source code.
The file `QtCore.toml` in the SIP folder for package PyQt5 module QtCore mentions the requirement version of sip build system.
```
sip-version = "6.7.12"
sip-abi-version = "12.13"
```

But the command line tool `sip` which is invoked when building, has definitely been removed in `SIP v6`, as it's mot mentioned in the sip repository.
[sip - command_line_tools.rst](https://www.riverbankcomputing.com/hg/sip/file/ac5e2889ba89/doc/command_line_tools.rst)

The relevant code is here:
https://github.com/ros-visualization/python_qt_binding/blob/e0f21d1cf19e9581406e83069fc462548c73647a/cmake/sip_configure.py#L146-L156

If I install the sip with version 4.19.25 and rebuild the RViz I got a `syntax error` from sip.
```
sip: {PYTHON_ENV_PATH}/lib/python3.9/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
subprocess.CalledProcessError: Command '['{PYTHON_ENV_PATH}/bin/sip', '-c', '{CATKIN_WORKSPACE}/build/rviz/sip/rviz_sip', '-b', '{CATKIN_WORKSPACE}/build/rviz/sip/rviz_sip/pyqtscripting.sbf', '-I', '{PYTHON_ENV_PATH}/lib/python3.9/site-packages/PyQt5/bindings', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_2', '-t', 'WS_X11', 'rviz.sip']' returned non-zero exit status 1.
make[2]: *** [rviz/src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:95: rviz/sip/rviz_sip/Makefile] Error 1
```
Is it a problem about version compatibility or I did some steps wrong?

There's a relevant reference #1777

rhaschke, 2024-02-02:
I'm not sure how you try to build rviz. In CI, everything [works as expected](https://github.com/ros-visualization/rviz/actions/runs/6881321243/job/18717427990#step:4:404). Hence, I assume, you are building rviz in the wrong way. Please describe in detail, which commands you execute. Maybe you missed to install all prerequisites?

`/usr/bin/sip` definitely exists in Ubuntu 20.04: it's part of package `sip-dev` and it's version is 4.19.21.

NumbNutN, 2024-02-03:
Thanks for replying! This is what I've done:
source tree like:
```
template
├── src
     └──rviz
```
```
cd template/src
catkin_init_workspace
cd ..
catkin_make
```

My output from stdout and stderr when building:
https://github.com/NumbNutN/cloud/blob/e66441b7795bc4d9b6539eea823dc999f12c24fd/rviz_build_info.log#L379-L388

I use a conda environment and `sip` is invoked the one from `pip` installed package when building
another attempt to run `/usr/bin/sip` failed:
```
$ /usr/bin/sip -c {CATKIN_WORKSPACE}/build/rviz/sip/rviz_sip -b {CATKIN_WORKSPACE}/build/rviz/sip/rviz_sip/pyqtscripting.sbf -I /home/numbnut/miniconda3/envs/rviz39/lib/python3.9/site-packages/PyQt5/bindings -w -n PyQt5.sip -t Qt_5_15_2 -t WS_X11 {CATKIN_WORKSPACE}/src/rviz/src/python_bindings/sip/rviz.sip
sip: /home/numbnut/miniconda3/envs/rviz39/lib/python3.9/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error

$ /usr/bin/sip -V
4.19.21
```

rhaschke, 2024-02-03:
The problem is you conda environment. You should not use ROS within conda environments except you are an absolute expert...

NumbNutN, 2024-02-03:
Yes.. it was built correctly after I deactivated conda. Thanks!

closed 2025-05-02 by rhaschke