ros-visualization/rviz #1816: ROS Noetic RVIZ Install From Source

RP-ND, 2024-02-11:
### Your environment
* OS Version: e.g. Ubuntu 22.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz: See console output:

Command Output:

> -- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: filesystem program_options system thread 
> -- Checking for modules 'OGRE;OGRE-Overlay'
> --   Found OGRE, version 1.9.0
> --   Found OGRE-Overlay, version 1.9.0
> -- OGRE_INCLUDE_DIRS=/usr/include/OGRE/Overlay;/usr/include/OGRE
> -- OGRE_LIBRARIES=/usr/lib/aarch64-linux-gnu/libOgreOverlay.so;/usr/lib/aarch64-linux-gnu/libOgreMain.so;/usr/lib/aarch64-linux-gnu/libpthread.a
> -- OGRE_PLUGIN_DIR=/usr/lib/aarch64-linux-gnu/OGRE-1.9.0
> -- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
> -- Found Qt 5.15.3
> -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
> /home/foo/ros_noetic_base_2204/catkin_ws/src/catkin/cmake/interrogate_setup_dot_py.py:43: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
>   import distutils.core
> -- rviz: 0 messages, 1 services
> -- Found X11: /usr/include   
> -- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so
> -- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so - found
> -- Looking for gethostbyname
> -- Looking for gethostbyname - found
> -- Looking for connect
> -- Looking for connect - found
> -- Looking for remove
> -- Looking for remove - found
> -- Looking for shmat
> -- Looking for shmat - found
> -- Looking for IceConnectionNumber in ICE
> -- Looking for IceConnectionNumber in ICE - found
> -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
> -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
> -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
> -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
> -- Performing Test COMPILER_HAS_DEPRECATED_ATTR
> -- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
> -- making version 1.14.20.
> No display, will not configure tests for rviz/stl_loader
> -- Configuring incomplete, errors occurred!
> See also "/home/foo/ros_noetic_base_2204/catkin_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/home/foo/ros_noetic_base_2204/catkin_ws/build/CMakeFiles/CMakeError.log".
> Invoking "cmake" failed

* If source build, which git commit?

Git Commit: 9911c32

* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: en_US.UTF-8 

Current issue is that the math library seems to return and can not find "trunc", here is the output of the CMakeError.log:

> Determining if the function trunc exists failed with the following output:
> Change Dir: /home/foo/ros_noetic_base_2204/catkin_ws/build/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_07aae/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_07aae.dir/build.make CMakeFiles/cmTC_0>
> gmake[1]: Entering directory '/home/foo/ros_noetic_base_2204/catkin_ws/build/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_07aae.dir/CheckFunctionExists.c.o
> /usr/bin/cc   -DCHECK_FUNCTION_EXISTS=trunc -o CMakeFiles/cmTC_07aae.dir/CheckFunctionExists.c.o -c /usr/share/cmake-3.22/Modules/CheckFunct>
> <command-line>: warning: conflicting types for built-in function ‘trunc’; expected ‘double(double)’ [-Wbuiltin-declaration-mismatch]
> /usr/share/cmake-3.22/Modules/CheckFunctionExists.c:7:3: note: in expansion of macro ‘CHECK_FUNCTION_EXISTS’
>     7 |   CHECK_FUNCTION_EXISTS(void);
>       |   ^~~~~~~~~~~~~~~~~~~~~
> /usr/share/cmake-3.22/Modules/CheckFunctionExists.c:1:1: note: ‘trunc’ is declared in header ‘<math.h>’
>   +++ |+#include <math.h>
>     1 | #ifdef CHECK_FUNCTION_EXISTS
> Linking C executable cmTC_07aae
> /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_07aae.dir/link.txt --verbose=1
> /usr/bin/cc  -DCHECK_FUNCTION_EXISTS=trunc -rdynamic CMakeFiles/cmTC_07aae.dir/CheckFunctionExists.c.o -o cmTC_07aae
> /usr/bin/ld: CMakeFiles/cmTC_07aae.dir/CheckFunctionExists.c.o: in function `main':
> CheckFunctionExists.c:(.text+0x10): undefined reference to `trunc'
> collect2: error: ld returned 1 exit status
> gmake[1]: *** [CMakeFiles/cmTC_07aae.dir/build.make:99: cmTC_07aae] Error 1
> gmake[1]: Leaving directory '/home/foo/ros_noetic_base_2204/catkin_ws/build/CMakeFiles/CMakeTmp'
> gmake: *** [Makefile:127: cmTC_07aae/fast] Error 2

Please let me know if any further information is needed. Thank you.

rhaschke, 2024-02-11:
I'm not sure what you are doing. ROS Noetic is targeted to Ubuntu 20.04. However, you state Ubuntu 22.04.
If you are looking for a Noetic-based release into Ubunto 22.04, have a look at https://ros.packages.techfak.net and https://github.com/ubi-agni/ros-builder-action.

RP-ND, 2024-02-11:
I installed from source Noetic on LTS 22.04 as there is a specific package I need for my use case. It has a use in rviz so I wanted to build that from source as well. All upstream packages were already built from source, it is just this CMAKE error gets thrown because of some problem with a linker to the math library and I am not sure where in the CMakeLists.txt this occurs at.

rhaschke, 2024-02-11:
I have no idea, where this error comes from and I can't reproduce it.

RP-ND, 2024-02-11:
Very well then, I will continue to debug this issue and update this when I see what has occurred wrong. I will close it otherwise.

closed 2024-02-21 by RP-ND