ros-visualization/rviz #1818: status error: points should not be empty

Qhijane, 2024-04-03:
when I running the code of FUEL, which I have facing an issue. and I do not know how do I can fix it. the error as following the picture below.
[image]
I am using Ubuntu 20.04. I am a greenman and have just started to come into contact with this software. plz help me to solve this error or plz give me some advices to fix this bug. I'll appreciate you to help me.thank you so much, best wishes!

closed 2024-04-03 by rhaschke

Qhijane, 2024-04-03:
> The error originates from a marker message, which - unexpectedly - has an empty `points` field.
> You should file an issue with the FUEL software package to get this fixed.

hello, thank you to reply. the link is the FUEL code address: https://github.com/HKUST-Aerial-Robotics/FUEL.git. I read the code and problems of the relevant packages, and I still haven't found a good way to solve this problem. So I am looking for help from teachers/scholars here.I would like to apologize to all teachers and scholars if I have disturbed you.

rhaschke, 2024-04-03:
You don't need to fix the problem yourself. Just file an issue at FUEL's github site.