ros-visualization/rviz #1819: Could not load resource [%s]: Unable to open file "%s"

PhilVince96, 2024-04-04:
I really did search every forum possible for this problem and I am confident, that there is every library installed that is needed and that the path is correct (I have already tried every possible path).

When I load the robot model into rviz it throws this error in the output:
```
Could not load resource [package://wam_description/urdf/meshes/wam_stand.stl]: Unable to open file "package://wam_description/urdf/meshes/wam_stand.stl".
```
for multiple files. I did try absolute paths, the file:/// option, $(find ...), etc. always the same output.

rat_console output:
```
Node: /wam/rviz
Time: 17:45:16.118340000 (2024-04-04)
Severity: Error
Published Topics: /clicked_point, /initialpose, /move_base_simple/goal, /rosout

Could not load resource [package://wam_description/urdf/meshes/wam_stand.stl]: Unable to open file "package://wam_description/urdf/meshes/wam_stand.stl".

Location:
/Users/fischert/mambaforge/envs/devenv/conda-bld/ros-noetic-rviz-0_1706257461249/work/ros-noetic-rviz/src/work/src/rviz/mesh_loader.cpp:loadMeshFromResource:658
```

The ROS_ERROR is thrown in row 658 of mesh_loader.cpp in the loadMeshFromResource function.
I do not get why the "resource_path" fails the assertions for the .stl extension and my only guess is that the os is the problem, because I know that this setup works on ubuntu. The .stl is also correct, I checked that with rosin xacro.

### Your environment
* OS Version: macOS 14.4.1 23E224 arm64
* ROS Distro: Noetic
* ROS Version: 1.16.0
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1712245369.622606000]: rviz version 1.14.20
[ INFO] [1712245369.622652000]: compiled against Qt version 5.15.8
[ INFO] [1712245369.622656000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1712245369.635788000]: Forcing OpenGl version 0.
[ INFO] [1712245370.037545000]: Stereo is NOT SUPPORTED
[ INFO] [1712245370.037596000]: OpenGL device: Apple M1
[ INFO] [1712245370.037602000]: OpenGl version: 2.1 (GLSL 1.2).
```

* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: en_US.UTF-8
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`! ✅ did not change the behaviour

rhaschke, 2024-04-04:
My best guess is to check the version of assimp used. Ensure that you build against the same version as used in Ubuntu.

PhilVince96, 2024-04-04:
Which build do you mean exactly? Because I use robostack on the macOS as well as on the Ubuntu system. But on the latter assimp is version 5.2.5 and on the macOS system it is version 5.3.1. But with assimp version 5.2.5 on the macOS system I still get the exact same error.

rhaschke, 2024-04-04:
If switching the version of assimp yields the same result, I don't have any further ideas. Sorry.