ros-visualization/rviz #1821: python Rviz binding not running on GPU on Orange pi 5 (RK3588S GPU Mali 610)

programmeddeath1, 2024-04-26:
Hiii

I am trying to run a pyqt application which uses RVIZ.

When starting the app which loads the 3D model on start it gives the following warnings -
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
[ INFO] [1714028941.896605237]: Stereo is NOT SUPPORTED
[ INFO] [1714028941.896694679]: OpenGL device: llvmpipe (LLVM 12.0.0, 128 bits)
[ INFO] [1714028941.896733974]: OpenGl version: 3.1 (GLSL 1.4).
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip 
The app does load properly, the pyqt application itself does run smoothly, but when i run certain motions on the simulator, it seems to be lagging a lot. I ran LIBGL_DEBUG=verbose python3 main.py and it gives the following detailed error - [file](https://github.com/programmeddeath1/webhost/blob/master/opi_rockhip_driver_error)

The code where the warning shows up is as below -

        self.RVIZ_FRAME = rviz.VisualizationFrame()
        self.reader = rviz.YamlConfigReader()
        self.rviz_config = rviz.Config()
        self.reader.readFile( self.rviz_config, robot_simulation_file )

        print("Starts here")
        self.framerviz = self.RVIZ_FRAME
        self.framerviz.setSplashPath( "" )
        self.framerviz.initialize()
        print("initialized")
        self.framerviz.load( self.rviz_config )
        self.setWindowTitle( self.rviz_config.mapGetChild( "Title" ).getValue() )
        print("Loaded")
What do i need to enable the gpu driver usage or improve the rendering, any help would be appreciated.

### Your environment
* OS Version: e.g. Ubuntu 20.04 docker
* ROS Distro: Noetic
* RViz, Qt, OpenGl version as printed by rviz:
    ```
    paste your console output here
    ```
* If source build, which git commit?
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`:
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[ INFO] [1714099555.948648663]: rviz version 1.14.20
[ INFO] [1714099555.948717786]: compiled against Qt version 5.12.8
[ INFO] [1714099555.948752494]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1714099555.970871974]: Forcing OpenGl version 0.
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
[ INFO] [1714099557.872495589]: Stereo is NOT SUPPORTED
[ INFO] [1714099557.872592420]: OpenGL device: llvmpipe (LLVM 12.0.0, 128 bits)
[ INFO] [1714099557.872623336]: OpenGl version: 3.1 (GLSL 1.4).
```
Any help to make rviz read the GPU would be helpful!

rhaschke, 2024-04-26:
As far as I understand, the rendering works, but is slow, which means that OpenGL is using software rendering only (also indicated by OpenGL device: llvmpipe).
Can you run other OpenGL apps, e.g. `glxgears` from the `mesa-utils` package?

programmeddeath1, 2024-04-26:
I ran glxgears, it opens and shows around 500FPS avg. but the above warnings remain 
```
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
2139 frames in 5.0 seconds = 427.701 FPS
2660 frames in 5.0 seconds = 531.885 FPS
2693 frames in 5.0 seconds = 538.575 FPS
2568 frames in 5.0 seconds = 513.598 FPS
```

The detailed error shown during rviz python binding load  is https://github.com/programmeddeath1/webhost/blob/master/libgl_error_rviz.txt

The orangepi sbc has pinfrost drivers for the mali G610 GPU

I am drawing certain components on the rviz urdf of a delta robot during run time in a parallel thread, the simulation works very smoothly on my laptop. Could there be any optimizations needed to be done for ARM devices for running the simulation?

How do I make rviz know which gpu drivers to load?

rhaschke, 2024-04-26:
This is not a problem of rviz, but of your driver configuration. rviz doesn't load low-level hardware drivers, it just interacts with those drivers through OpenGL. Thus, if you have issues with OpenGL apps, like glxgears as well, first fix your driver setup. There are some issue reports available on the web, e.g.
https://askubuntu.com/questions/1493949/ubuntu-20-04-image-given-by-ornangepi-giving-error-while-running-rviz

Closing here, as this is not an rviz issue.

closed 2024-04-26 by rhaschke

rhaschke, 2024-04-26:
Your problem isn't rviz, but the GPU driver. Hence, you will face similar problems with rviz2.
I don't know the state of the python bindings of rviz2. Sorry.

programmeddeath1, 2024-04-26:
no actually, I am running on a docker of 20.04 inside ubuntu 22.04. The orangepi sbc has versions 22.04 and 24.04 released with proper gpu driver support. There I ran glxgears also outside the docker, they work properly and are synchronized with the monitor frame rate and do not give the above rockchip driver error.

So in their 22.04 os, glxgears loads the drivers properly  - ```LIBGL_DEBUG=verbose glxgears
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/greenox/.drirc: No such file or directory.
libGL: using driver rockchip for 4
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/greenox/.drirc: No such file or directory.
libGL: MESA-LOADER: device is not located on the PCI bus
libGL: using driver rockchip for 4
libGL: MESA-LOADER: dlopen(/usr/lib/aarch64-linux-gnu/dri/rockchip_dri.so)
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/greenox/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/greenox/.drirc: No such file or directory.
drmOpenDevice: node name is /dev/dri/renderD128
drmOpenDevice: open result is 6, (OK)
drmOpenDevice: node name is /dev/dri/renderD129
drmOpenDevice: open result is 6, (OK)
drmOpenDevice: node name is /dev/dri/renderD130
drmOpenDevice: node name is /dev/dri/renderD131
drmOpenDevice: node name is /dev/dri/renderD132
drmOpenDevice: node name is /dev/dri/renderD133
drmOpenDevice: node name is /dev/dri/renderD134
drmOpenDevice: node name is /dev/dri/renderD135
drmOpenDevice: node name is /dev/dri/renderD136
drmOpenDevice: node name is /dev/dri/renderD137
drmOpenDevice: node name is /dev/dri/renderD138
drmOpenDevice: node name is /dev/dri/renderD139
drmOpenDevice: node name is /dev/dri/renderD140
drmOpenDevice: node name is /dev/dri/renderD141
drmOpenDevice: node name is /dev/dri/renderD142
drmOpenDevice: node name is /dev/dri/renderD143
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/greenox/.drirc: No such file or directory.
libGL: Using DRI3 for screen 0
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
304 frames in 5.0 seconds = 60.710 FPS
295 frames in 5.0 seconds = 58.919 FPS
300 frames in 5.0 seconds = 59.877 FPS
299 frames in 5.0 seconds = 59.737 FPS
300 frames in 5.0 seconds = 59.883 FPS

I had faced a lot of dependency issues trying to build ros1 from source on ubuntu 22 so dropped that idea for now,

Okay I understand will try and explore more, thank you for your guidance!

rhaschke, 2024-04-26:
Good, that the driver issues are resolved in Ubuntu 22.04. If ROS2 is an option, I suggest trying it first.
I have built ROS1 .deb packages for Ubuntu 22.04 successfully: https://ros.packages.techfak.net
Maybe you can draw some inspiration from my corresponding github action: https://github.com/ubi-agni/ros-builder-action

programmeddeath1, 2024-04-26:
That looks mighty helpful. I will give it a try once before going for ROS2, since a ROS2 migration of my entire codebase would be a longer task.