ros-visualization/rviz #1822: cannot import name 'librviz_sip'/librviz_shiboken from partially initialized module 'rviz' (most likely due to a circular import)

programmeddeath1, 2024-04-29:
I have built ros1 on ubuntu 22.04 following these steps - https://gist.github.com/Meltwin/fe2c15a5d7e6a8795911907f627255e0 with rosdep and vcstool installed through your guide in https://ros.packages.techfak.net/

Rviz loads properly through rosrun rviz rviz. But when I try to import rviz from my pyqt application it fails with the following error -
```
Traceback (most recent call last):
  File "/home/greenpi/app/delta_app_master/main_delta_app/src/delta_gui/main.py", line 59, in <module>
    import rviz
  File "/home/greenpi/noetic_ws/devel_isolated/rviz/lib/python3/dist-packages/rviz/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 25, in <module>
ImportError: cannot import name 'librviz_sip' from partially initialized module 'rviz' (most likely due to a circular import) (/home/greenpi/noetic_ws/devel_isolated/rviz/lib/python3/dist-packages/rviz/__init__.py)
```
I tried using pyside, but it also fails with the same error for librviz_shiboken.

Can you guide me as to what could be wrong with my build or how I can workaround this issue? Is the python_binding not built properly?

### environment
* OS Version: Ubuntu 22.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
[ INFO] [1714366672.168074574]: rviz version 1.14.21
[ INFO] [1714366672.168175783]: compiled against Qt version 5.15.3
[ INFO] [1714366672.168188033]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1714366672.185101753]: Forcing OpenGl version 0.
[ INFO] [1714366672.664077381]: Stereo is NOT SUPPORTED
[ INFO] [1714366672.664220006]: OpenGL device: Mali-G610 (Panfrost)
[ INFO] [1714366672.664255881]: OpenGl version: 3.3 (GLSL 3.3) limited to GLSL 1.4 on Mesa system.
    ```
* git commit - rviz-release-release-noetic-rviz-1.14.21-1
* System locale - en_IN:

rhaschke, 2024-04-29:
If you found https://ros.packages.techfak.net, I don't understand why you built ROS from source. This package repo provides pre-compiled debs for most ROS1 packages.
However, I checked with my ROS One distro: rviz doesn't seem to compile (or at least install) the python bindings. Need to check later, why.

programmeddeath1, 2024-04-29:
Hi as I had mentioned in an earlier question I had raised where rviz was not able to access the GPU, the device I am running on is arm64 based and the pre-compiled debs are for either x86 or amd64 and hence it showed it is not installable in my SBC device.

I observed that the sip and shiboken folders inside rviz are not getting compiled even though CmakeLists.txt is defined. It is thus not creating Librviz_sip.so files compared to my ros-noetic installation on my laptop in ubuntu 20.04.

rhaschke, 2024-04-29:
I looked into this and noticed that python bindings were accidentally disabled in #1813 for no valid reason. Fixed now.

closed 2024-04-29 by rhaschke