ros-visualization/rviz #1825: After PR #1808: Segfault when loading RViz config with RobotModel display using non-default parameter name

peci1, 2024-05-09:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```
[ INFO] [1715256147.602104886] [ros.rviz]: rviz version 1.14.22
[ INFO] [1715256147.602171601] [ros.rviz]: compiled against Qt version 5.12.8
[ INFO] [1715256147.602181139] [ros.rviz]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1715256147.615263427] [ros.rviz]: Forcing OpenGl version 0.
[ INFO] [1715256147.923097661] [ros.rviz]: Stereo is NOT SUPPORTED
[ INFO] [1715256147.923240298] [ros.rviz]: OpenGL device: AMD Radeon Graphics (radeonsi, renoir, LLVM 15.0.7, DRM 3.57, 6.8.0-060800-generic)
[ INFO] [1715256147.923289680] [ros.rviz]: OpenGl version: 4,6 (GLSL 4,6) limited to GLSL 1.4 on Mesa system.
```
* If source build, which git commit? Tag 1.14.22
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: cs_CZ.UTF-8 
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`! - no effect

### Description

Commit df72fce669683e01a67156bec3f847f6449b30a7 from #1808 broke loading rviz with a config that contains RobotModel display configured for non-default parameter name (e.g. `robot_description2`). This only happens when `robot_description` also contains something, which is very weird, because I'd say that if the config only contains RobotModel with `robot_description2`, it should not touch `robot_description` at all. But it does and it leads to segfault.

### Repro

a.urdf:

```XML
<?xml version="1.0" ?>
<robot name="robot1">
  <link name="base" />
</robot>
```

c.urdf:

```XML
<?xml version="1.0" ?>
<robot name="robot2">
  <link name="base" />
  <joint name="base_to_base_inertia" type="fixed">
    <parent link="base"/>
    <child link="base_inertia"/>
  </joint>
  <link name="base_inertia" />
</robot>
```

b.rviz:

```YAML
Panels:
  - Class: rviz/Displays
    Name: Displays
Visualization Manager:
  Class: ""
  Displays:
    - Alpha: 1
      Class: rviz/RobotModel
      Collision Enabled: false
      Enabled: true
      Links:
        All Links Enabled: true
        Expand Joint Details: false
        Expand Link Details: false
        Expand Tree: false
        Link Tree Style: Links in Alphabetic Order
      Name: RobotModel
      Robot Description: robot_description2
      TF Prefix: ""
      Update Interval: 0
      Value: true
      Visual Enabled: true
  Enabled: true
  Name: root
  Value: true
Window Geometry:
  QMainWindow State: 000000ff00000000fd000000040000000000000156000002b0fc0200000008fb0000001200530065006c0065006300740069006f006e00000001e10000009b0000005c00fffffffb0000001e0054006f006f006c002000500072006f007000650072007400690065007302000001ed000001df00000185000000a3fb000000120056006900650077007300200054006f006f02000001df000002110000018500000122fb000000200054006f006f006c002000500072006f0070006500720074006900650073003203000002880000011d000002210000017afb000000100044006900730070006c006100790073010000003d000002b0000000c900fffffffb0000002000730065006c0065006300740069006f006e00200062007500660066006500720200000138000000aa0000023a00000294fb00000014005700690064006500530074006500720065006f02000000e6000000d2000003ee0000030bfb0000000c004b0069006e0065006300740200000186000001060000030c00000261000000010000010f000002c4fc0200000003fb0000001e0054006f006f006c002000500072006f00700065007200740069006500730100000041000000780000000000000000fb0000000a005600690065007700730000000028000002c4000000a400fffffffb0000001200530065006c0065006300740069006f006e010000025a000000b200000000000000000000000200000490000000a9fc0100000001fb0000000a00560069006500770073030000004e00000080000002e10000019700000003000004b00000003efc0100000002fb0000000800540069006d00650100000000000004b0000003bc00fffffffb0000000800540069006d0065010000000000000450000000000000000000000354000002b000000004000000040000000800000008fc0000000100000002000000010000000a0054006f006f006c00730100000000ffffffff0000000000000000
```

bug.launch:

```XML
<launch>
    <arg name="bug" default="true" />
    <param name="robot_description" textfile="$(dirname)/a.urdf" if="$(arg bug)" />
    <param name="robot_description2" textfile="$(dirname)/c.urdf" />
    <node name="rviz" pkg="rviz" type="rviz" args="-d $(dirname)/b.rviz" />
</launch>
```

When you launch `roslaunch ./bug.launch`, rviz 1.14.22 segfaults. When you launch `roslaunch ./bug.launch bug:=false`, it launches correctly (that corresponds to not setting `robot_description`). Be sure to let roslaunch create its own rosmaster when doing these tests. 

### GDB backtrace

```
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007ffff6eb1a2e in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt
#0  0x00007ffff6eb1a2e in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff7d0e9ca in rviz::PropertyTreeModel::propertyHiddenChanged(rviz::Property const*) (this=0x0, _t1=0x555556cea890)
    at /tmp/ws/build/rviz/src/rviz/rviz_autogen/BW4I5L4EMK/moc_property_tree_model.cpp:169
#2  0x00007ffff7def741 in rviz::PropertyTreeModel::emitPropertyHiddenChanged(rviz::Property const*) (this=0x0, property=0x555556cea890) at /tmp/ws/src/rviz/src/rviz/properties/property_tree_model.h:137
#3  0x00007ffff7dee4fb in rviz::Property::<lambda()>::operator()(void) const (__closure=0x555556c20f40) at /tmp/ws/src/rviz/src/rviz/properties/property.cpp:401
#4  0x00007ffff7def5f4 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, rviz::Property::setModel(rviz::PropertyTreeModel*)::<lambda()> >::call(rviz::Property::<lambda()> &, void **)
    (f=..., arg=0x7fffffffc8d0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:146
#5  0x00007ffff7def5c6 in QtPrivate::Functor<rviz::Property::setModel(rviz::PropertyTreeModel*)::<lambda()>, 0>::call<QtPrivate::List<>, void>(rviz::Property::<lambda()> &, void *, void **)
    (f=..., arg=0x7fffffffc8d0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:256
#6  0x00007ffff7def594 in QtPrivate::QFunctorSlotObject<rviz::Property::setModel(rviz::PropertyTreeModel*)::<lambda()>, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) (which=1, this_=0x555556c20f30, r=0x555556c28e30, a=0x7fffffffc8d0, ret=0x0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:439
#7  0x00007ffff6ebf486 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff6eb2bc5 in QObject::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff736aa66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff73740f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff6e8680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff6edd780 in QTimerInfoList::activateTimers() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff6ede06c in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff414217d in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff4142400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff41424a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff6ede4b4 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff7e65e99 in rviz::VisualizationFrame::initialize(QString const&) (this=0x5555557f3800, display_config_file=...) at /tmp/ws/src/rviz/src/rviz/visualization_frame.cpp:367
#19 0x00007ffff7e85723 in rviz::VisualizerApp::init(int, char**) (this=0x7fffffffd3e0, argc=5, argv=0x7fffffffd568) at /tmp/ws/src/rviz/src/rviz/visualizer_app.cpp:213
#20 0x000055555555e6a3 in main(int, char**) (argc=5, argv=0x7fffffffd568) at /tmp/ws/src/rviz/src/rviz/main.cpp:39
```

### Analysis

When I revert commit df72fce669683e01a67156bec3f847f6449b30a7, the bug does not manifest.

If I start an empty RViz and add the RobotModel display to it, no segfaults happen. But when I save it and rerun RViz with this config, and `robot_description` is set, the segfault happens.

rhaschke, 2024-05-09:
Fixed via e0ec5f6b5d4b06042c51501c20fb6f6a1c5d4f0d

peci1, 2024-05-09:
Oh, great, thanks a lot for the quick action, @rhaschke ! I really didn't know what the culprit could be...

rhaschke, 2024-05-12:
Thanks a lot, Martin, for your excellent bug report. That helped a lot to pinpoint the issue quickly.

Achllle, 2024-05-17:
I am still getting this issue (or at least very similar) on latest `noetic-devel` (2db2c5394a4f1fcf5fecfb8d0952d4f913c04aee). 

Renaming a `rviz/RobotModel` display and relaunching causes RViz to crash. Reverting to version 1.14.20 makes it work again with the issue being introduced in 1.14.21. I cannot reproduce with the example provided by @peci1 so this might be a different issue. I will try to create a minimally reproducible example and post here or create a new issue but my guess is that I've found an edge case that's not solved by @rhaschke's fix.

rhaschke, 2024-05-17:
@Achllle: I cannot reproduce your issue. As you complain about renaming the display (from `RobotModel`) and not about renaming the topic, I guess that's a different issue. Please open a new issue report and provide a detailed description how to reproduce the error. If you have an rviz config reproducing it, please upload that file.

PR #1831 (2024-05-21): Fix #1825/#1830: segfault when hiding deleted properties
When a hidden property was configured to process `propertyHiddenChanged` and got deleted in between, this was causing a segfault. We need to protect the slot/functor from being called when the property is deleted.

Fixes #1830.
commit 89a541f1 (2024-05-21): Fix #1825/#1830: segfault when hiding deleted properties (#1831)