ros-visualization/rviz #1826: Map Display lags behind due to Qt::QueuedConnection

MichaelGrupp, 2024-05-13:
After the recent upgrade from 1.14.20 to 1.14.23, we observed glitches when rendering dynamic maps such as costmaps.

I tracked the problem down to this commit: https://github.com/ros-visualization/rviz/commit/93ac44b26babb298046694efe78f8f986f7219e9 (PR https://github.com/ros-visualization/rviz/pull/1793)

This "hack" seems to cause a delay in map rendering that leads to a broken map display.

### With QueuedConnection

Using an unmodified 1.14.23 from the ROS noetic package distro. The costmap is jumping while moving. Note that the costmap frame_id is in a fixed frame.

https://github.com/ros-visualization/rviz/assets/8985495/6e592999-bd06-4403-afaa-9803da4773c7

### Without QueuedConnection

I cloned RViz from source at 1.14.23, removed that `Qt::QueuedConnection` flag, and built it. Using this patched RViz, the costmap is rendered at the correct spot again without jumping:

https://github.com/ros-visualization/rviz/assets/8985495/b0494338-cfb6-4262-9a3f-1a53f9a75c76

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[${node} INFO 1715596133.955965167]: rviz version 1.14.23
[${node} INFO 1715596133.956069614]: compiled against Qt version 5.12.8
[${node} INFO 1715596133.956126962]: compiled against OGRE version 1.9.0 (Ghadamon)
[/rviz_1715596402254328779 INFO 1715596402.270469085]: Forcing OpenGl version 0.
[/rviz_1715596402254328779 INFO 1715596402.742861634, 1715596391.783960104]: Stereo is NOT SUPPORTED
[/rviz_1715596402254328779 INFO 1715596402.743000877, 1715596391.783960104]: OpenGL device: AMD RENOIR (DRM 3.44.0, 5.16.15-051615-generic, LLVM 12.0.0)
[/rviz_1715596402254328779 INFO 1715596402.743048266, 1715596391.783960104]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
```
* If source build, which git commit? -
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: `en_US.utf8` (`LANG=C rviz` doesn't change anything here)
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

MichaelGrupp, 2024-05-13:
PR with code changes for my workaround: https://github.com/ros-visualization/rviz/pull/1827

rhaschke, 2024-05-13:
Thanks for reporting this issue. Could you please provide a bagfile with your map data to experiment with the issue?

rhaschke, 2024-05-14:
Thanks for providing the bag file. I can reproduce the glitches with that file. However, a simple test example ([map_test.py](https://github.com/ros-visualization/rviz/blob/f35dee1b8ebd0dccd225f2b48832dc0036dc6bfc/src/test/map_test.py)) doesn't exhibit the issue. Thus I don't yet understand, where your particular issue comes from.
What version of the map is shown for the fraction of a second?
Could you please experiment with this test map and try to reproduce your issue with it?

MichaelGrupp, 2024-05-14:
The issue seems to be visible with the following conditions:

* grid is moved
* content of grid changes

With the queued connection enabled, the previous grid gets first moved to the new position. Then the content gets updated with a short delay. It's not the grid moving to the wrong position, this is easier to see when replaying the videos with slower playback speed.

Here's a modified version of your script that makes it a bit better visible, it uses the RViz logo image from this repository:
https://github.com/magazino/rviz/commit/0bc892642811302511a7a26e7d51dc07ccd0bc28

It looks like in the video below in RViz 1.14.23. Also here I'd recommend to reduce playback speed, but it's also visible in normal speed.

https://github.com/ros-visualization/rviz/assets/8985495/c1d8dfa6-4160-4fbb-9376-88cc81bfe2af

With the modified source without queued connection:

https://github.com/ros-visualization/rviz/assets/8985495/d6938c48-c2e2-4b1b-8767-dc6e9cc9ea2a

PR #1828 (2024-05-15): Fix #1826: jitter in Map display updates
The queued connection update introduced in #1793 caused the map display to first update the pose of the map and then the map itself (in the next update cycle). The resulting jittering can be perfectly seen when throttling the rviz frame rate.

Updates to the visualization should be handled in update() only. Thus, now, if a map update is received, a flag is set to perform the costly map update. If that flag is not set, only the transform is updated.

Fixes #1826
commit 7e65fd29 (2024-05-15): Fix #1826: jitter in Map display updates (#1828)

The queued connection update introduced in #1793 caused the map display
to first update the pose of the map and then the map itself (in the next update cycle).
The resulting jittering can be perfectly seen when throttling the rviz frame rate.

Updates to the visualization should be handled in update() only.
Thus, now, if a map update is received, a flag is set to perform the costly map update.
If that flag is not set, only the transform is updated.
closed 2024-05-15 by rhaschke