ros-visualization/rviz #1829: After updating from 1.14.20 to 1.14.23 version, there is an interface incompatibility

tsukione, 2024-05-17:
After updating from 1.14.20 to 1.14.23 version, there is an interface incompatibility. It seem about rviz::EditableEnumProperty. could it be rollback?
```
[ERROR] [1715910267.480759600]: PluginlibFactory: The plugin for class 'rviz_plugin/ObjectArray' failed to load.  Error: Failed to load library /home/rviz_plugin/lib/librviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/rviz_plugin/lib//librviz_plugin.so: undefined symbol: _ZN4rviz20EditableEnumPropertyC1ERK7QStringS3_S3_PNS_8PropertyEPKcP7QObject)
```

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

tsukione, 2024-05-17:
or is there any way I can temporarily downgrade back to 1.14.20, using apt-get?

rhaschke, 2024-05-17:
The missing symbol corresponds to the following constructor:
https://github.com/ros-visualization/rviz/blob/fd566b96d54f6f2f003291dd99563396ee5809e4/src/rviz/properties/editable_enum_property.h#L47-L52

The Qt4-style signals have been replaced with modern Qt5-style signals. The constructor now looks as follows:
https://github.com/ros-visualization/rviz/blob/2db2c5394a4f1fcf5fecfb8d0952d4f913c04aee/src/rviz/properties/editable_enum_property.h#L52-L58

You need to adapt the construction and pass your `change_slot` as a method pointer instead of a string:
`EditableEnumProperty("name", "default", "description", parent, &MyClass::changed_slot, receiver);`

rhaschke, 2024-05-17:
Actually, it should suffice to recompile your plugin. I designed the new API to be backward-compatible. However, it is not ABI-compatible. So a recompile is required.

closed 2024-06-06 by rhaschke