ros-visualization/rviz #1832: Rviz version 1.14.23: librviz related crash after ubuntu (20.04) update

arbephoenix, 2024-05-23:
We are using ROS Noetic on Ubuntu 20.04 and have an application (GUI) that we developped and uses librviz.so 
All was working well for the last two year until the latest Ubuntu update broke our application.

The Rviz version after the ubuntu update is 1.14.23

Below is the segmentation fault we are seeing.
Any hints why we started seeing this after the latest ubuntu update?

(gdb) bt
#0  0x00007ffff71968b4 in QWidget::layout() const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff728eb6d in QDockWidget::titleBarWidget() const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff7d8635f in rviz::PanelDockWidget::addMaximizeButton() () at /opt/ros/noetic/lib/librviz.so
#3  0x00007fff34786fef in rviz::ImageDisplay::onInitialize() () at /opt/ros/noetic/lib/librviz_default_plugin.so
#4  0x00007ffff7d102a5 in rviz::Display::initialize(rviz::DisplayContext*) () at /opt/ros/noetic/lib/librviz.so
#5  0x00007ffff7dfb058 in rviz::VisualizationManager::addDisplay(rviz::Display*, bool) () at /opt/ros/noetic/lib/librviz.so
#6  0x00007ffff7dfb7e8 in rviz::VisualizationManager::createDisplay(QString const&, QString const&, bool) () at /opt/ros/noetic/lib/librviz.so
#7  0x00005555556908a3 in viewpanel::enableCamera(bool) (this=0x555558d1ec00, isUSB=false)
    at /home/eyalr/arbe-gitlab/arberos_1.3/src/arbe_phoenix_radar_driver/arbe_gui/src/arbe_gui_main/viewpanel.cpp:4445
#8  0x000055555561a253 in MainWindow::enableCamera() (this=0x7fffffffb8d0)
    at /home/eyalr/arbe-gitlab/arberos_1.3/src/arbe_phoenix_radar_driver/arbe_gui/src/arbe_gui_main/mainwindow.cpp:1240
#9  0x0000555555615c40 in MainWindow::MainWindow() (this=0x7fffffffb8d0)
    at /home/eyalr/arbe-gitlab/arberos_1.3/src/arbe_phoenix_radar_driver/arbe_gui/src/arbe_gui_main/mainwindow.cpp:659
#10 0x00005555555a456c in main(int, char**) (argc=1, argv=0x7fffffffc308)
    at /home/eyalr/arbe-gitlab/arberos_1.3/src/arbe_phoenix_radar_driver/arbe_gui/src/arbe_gui_main/main.cpp:1970
(gdb)

rhaschke, 2024-05-23:
Could you please try the latest version 1.14.25, building from source or from the [ROS `testing` repo](http://wiki.ros.org/TestingRepository)?

rhaschke, 2024-05-23:
Unfortunately, there was a bug introduced, which was not noticed during the testing phase.

arbephoenix, 2024-05-24:
Thank you for confirming!
Do you have any estimation of when 1.14.25 is going to hit the next Ubuntu update.

We will try building from sources to confirm the fix

Thanks again for you prompt response!

rhaschke, 2024-05-24:
Usually, sync intervals are within 1-2 months. Until then, I suggest to install the debian package from the [ROS testing repo](http://wiki.ros.org/TestingRepository).

arbephoenix, 2024-05-25:
Thank you!
I updated ros-noetic-rviz from ros-testing to 1.14.25 using the suggested method and the issue is resolved.

Have a great weekend,
Eyal

closed 2024-05-25 by rhaschke