ros-visualization/rviz #1833: Add background overlay under text view facing

ClementLeBihan, 2024-05-28:
Hi,
I am trying to add a background rectangle behind the text view facing to make it more readable. Do you have any advice to do so ? 
Best,
Clément

ClementLeBihan, 2024-05-28:
I do not know what is the proper way using Ogre to display Text with an overlay. If this is already possible with Ogre or if I need to create a rectangle behind the text.

rhaschke, 2024-05-28:
I don't know this by heart too. You will need to consult the docs.

ClementLeBihan, 2024-05-29:
I can see a mpBackgroundMaterial variable in movable_text header but never used. Is it here to do what I want ?

ClementLeBihan, 2024-05-30:
I'm trying to understand how the movable_text. 
It use vertexData that are filled with POS and Texture cordinates for each letter of the text, refering the material. 
Do you guys know if it's possible to add extra vertex that doesn't contain texture cordinates to draw a rectangle ? To mix vertex format in a certain way (letters vertex and triangle vertex). I can't make it work on my side.

ClementLeBihan, 2024-06-05:
I succeed to make it work adding new vertex corresponding to the background rectangle I wanted to display. I didn't find a way to avoid using POS + TEXTURE, so I found a trick where I choose the middle of a letter to display a full black rectangle ...

closed 2024-06-05 by ClementLeBihan