ros-visualization/rviz #1834: Rviz version 1.14.25: still another librviz related crash

arbephoenix, 2024-05-28:
with regards to the closed issue "Rviz version 1.14.23: librviz related crash after ubuntu (20.04) update #1832"

Apparently after testing some more, we have actually seen two seperate crash types in 1.14.23 that are not seen with 1.14.20.
One was fixed with 1.14.25 buty the other one is still there.
See the description of both below.
Issue 1 is solved with 1.14.25 but issue 2 still persists

1. This crash was indeed solved with 1.14.25 (was crashing with 1.14.23)

(gdb) bt
#0  0x00007ffff64b1a2e in QMetaObject::activate(QObject*, int, int, void**) ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff7d0e90a in rviz::PropertyTreeModel::propertyHiddenChanged(rviz::Property const*) () at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#2  0x00007ffff7deef6b in rviz::PropertyTreeModel::emitPropertyHiddenChanged(rviz::Property const*) ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#3  0x00007ffff7dedd25 in rviz::Property::setModel(rviz::PropertyTreeModel*)::{lambda()#1}::operator()() const ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#4  0x00007ffff7deee1e in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, rviz::Property::setModel(rviz::PropertyTreeModel*)::{lambda()#1}>::call({lambda()#1}&, void**) ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#5  0x00007ffff7deedf0 in void QtPrivate::Functor<rviz::Property::setModel(rviz::PropertyTreeModel*)::{lambda()#1}, 0>::call<QtPrivate::List<>, void>({lambda()#1}&, void*, {lambda()#1}&*) ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#6  0x00007ffff7deedbe in QtPrivate::QFunctorSlotObject<rviz::Property::setModel(rviz::PropertyTreeModel*)::{lambda()#1}, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#7  0x00007ffff64bf486 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#8  0x00007ffff64b2bc5 in QObject::event(QEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff6f6aa66 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff6f740f0 in QApplication::notify(QObject*, QEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff648680a in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff64dd780 in QTimerInfoList::activateTimers() ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff64de06c in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007fff8814217d in g_main_context_dispatch ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007fff88142400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007fff881424a3 in g_main_context_iteration ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff64de435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff64853ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff648d116 in QCoreApplication::exec() ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x0000555555599801 in main ()
(gdb) 

2. This issue is still there with 1.14.25

(gdb) bt
#0  0x00007ffff6f968b4 in QWidget::layout() const ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff708eb6d in QDockWidget::titleBarWidget() const ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff7dd67c3 in rviz::PanelDockWidget::addMaximizeButton() ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#3  0x00007fff119760fa in rviz::ImageDisplay::onInitialize() ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz_default_plugin.so
#4  0x00007ffff7d20ad5 in rviz::Display::initialize(rviz::DisplayContext*) ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#5  0x00007ffff7e7e139 in rviz::VisualizationManager::addDisplay(rviz::Display*, bool) () at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#6  0x00007ffff7e7e694 in rviz::VisualizationManager::createDisplay(QString const&, QString const&, bool) ()
    at /home/eyalr/arbe-gitlab/arberos_1.3/devel/lib/librviz.so
#7  0x0000555555636a87 in viewpanel::enableCamera(bool) ()
#8  0x0000555555604ab2 in MainWindow::MainWindow() ()
#9  0x0000555555599198 in main ()

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
rviz --version
[ INFO] [1716898124.335460451]: rviz version 1.14.23
[ INFO] [1716898124.335527302]: compiled against Qt version 5.12.8
[ INFO] [1716898124.335539388]: compiled against OGRE version 1.9.0 (Ghadamon)* 

Any suggestions?

Best Regards,
Eyal

arbephoenix, 2024-05-28:
Basically we are calling the following sequence from a custom GUI implemented in c++ that uses librviz
Attching the relevant snippet below.
The string title that is built below is "Front Camera"
nr equals 0
..
		std::string title = Radar_name[nr] + " Camera";
		Camera_[nr] = manager_->createDisplay("rviz/Image", title.c_str(), true);
..

The crash happens when the createDisplay API ic called

rhaschke, 2024-05-28:
In the rviz GUI those floating windows are DockWidgets, which have a title bar. In order to understand, what needs to be changed in the code, I do need a minimal example of your code to reproduce the issue.

rhaschke, 2024-05-28:
From the gdb backtrace, I guess that #1835 should fix your issue. Could you please verify?

arbephoenix, 2024-05-28:
> From the gdb backtrace, I guess that #1835 should fix your issue. Could you please verify?

Yes!
I have manually merged this fix into image_display.cpp in the rviz source code and it looks like it is sollving the issue:
https://github.com/ros-visualization/rviz/pull/1835/files

I assume this fix is just not merged yet into noetic_devel, right?

See the "working" screenshot:
[image]

Best Regards,
Eyal

closed 2024-05-28 by rhaschke

rhaschke, 2024-05-28:
Thanks for your swift reply. I merged the fixes into the noetic-devel branch.
I will wait with a new release a few more days. Maybe more issues pop up? 
I'm sorry for the trouble this latest updated caused. 
Please give me a reminder, if I didn't pushed a new release within 2 weeks.

commit fdcf656a (2024-05-28): Fix #1834: segfault on addMaximizeButton

... for image/camera displays if no window manager is instantiated