ros-visualization/rviz #1838: RViz Freezing with Autoware

dtonda8, 2024-08-05:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 22.04
* ROS Distro: Humble
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    (see launch.log file at the end)
    ```
* If source build, which git commit? (?) Followed the Default ROS2 installation (I think this is where I installed RViz), als using the [latest Autoware Source Installation](https://www.google.com/search?client=safari&rls=en&q=source+installation+autoware&ie=UTF-8&oe=UTF-8)
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: en_US.UTF-8
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

Rviz2 works find with an empty configuration. However, when following the Autoware Planning Simulation instructions, The centre display freezes the second time I try to interact with the 3D space:

https://github.com/user-attachments/assets/3ae6de2e-c1e9-4745-bc8b-fe9da17ee063

`$HOME/.ros/log` output:  [launch.log](https://github.com/user-attachments/files/16491604/launch.log)

rhaschke, 2024-08-05:
Please report this issue with the rviz2 repository: https://github.com/ros2/rviz

closed 2024-08-05 by rhaschke

dtonda8, 2024-08-05:
@rhaschke my bad! Thanks for the swift response!