ros-visualization/rviz #1841: What are the resources for 3D reconstruction + robot visualization?

guotong1988, 2024-11-14:
Thank you very much!!

rhaschke, 2024-11-14:
Can you please clarify you question? What resources are you referring to?

guotong1988, 2024-11-14:
Thank you for the response.

I am mainly looking for ways to align and visualize a 3D reconstructed model (prediction results) using tools like RViz for visualization.

rhaschke, 2024-11-14:
rviz is a visualization tool. It doesn't provide compute methods like alignment. There are numerous [display plugins](https://wiki.ros.org/rviz/DisplayTypes), including [markers](https://wiki.ros.org/rviz/DisplayTypes/Marker) (e.g. for meshes) and [point clouds](https://wiki.ros.org/rviz/DisplayTypes/PointCloud). And, of course, you can write your own plugin.

closed 2024-11-14 by guotong1988