ros-visualization/rviz #1842: Panels fail to render in docker.

miraculin1, 2024-11-17:
# Issue
- The view panel are not updated according to my move, and some times black. 
- The tile of time panel are black.
- the description text of Display panel not work properly.
- **When the panels resize or scroll cause some ghosting, but if move the panels out from default place(float or to another place) it will work as expected.**

*The pics attached below show the problem visually:*
[image]
[image]

## Reproduce

I'm running osrf/ros:noetic-desktop-full in docker. After installed nvidia-container-toolkit and run with below script:
```bash
docker run -it \
  --gpus all \
  --name $1 \
  --device=/dev/dri:/dev/dri \
  --network=host --ipc=host \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -e DISPLAY=$DISPLAY \
  --env="QT_X11_NO_MITSHM=1" \
  --env="NVIDIA_DRIVER_CAPABILITIERS=all" \
  --env="NVIDIA_VISABLE_DEVICES=all" \
  --device /dev:/dev \
  --privileged \
  -v ./mnt:/workspace \
  -e "TERM=xterm-256color" \
  --ulimit nofile=1024:524288 \
  noetic_dev:latest
```
Then run `xhost +local:` in host to enable docker's gui to display in host.

Finally run below command in docker cause the problem.
```bash
roscore & rviz
```

I've tried gazebo, gimp and glxgear in docker. They worked well.

### Your environment
* OS Version: Debian trixie
* RTX3060 laptop, using dynamic graphics.
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
    [ INFO] [1731839500.738132603]: rviz version 1.14.25
    [ INFO] [1731839500.738160960]: compiled against Qt version 5.12.8
    [ INFO] [1731839500.738184427]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1731839500.743302592]: Forcing OpenGl version 0.
    [ INFO] [1731839500.976761193]: Stereo is NOT SUPPORTED
    [ INFO] [1731839500.976814344]: OpenGL device: AMD RENOIR (DRM 3.59.0, 6.11.7-amd64, LLVM 12.0.0)
    [ INFO] [1731839500.976831456]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
    ```
* If source build, which git commit? -- Not source build.
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: `en_US.UTF-8` in both host and container.
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!: Same issue.

rhaschke, 2024-11-18:
I'm wondering why you use an nvidia container when your device is an AMD GPU.

miraculin1, 2024-11-18:
> I'm wondering why you use an nvidia container when your device is an AMD GPU.

Thanks for replying,
My laptop have a amdcpu integrate gpu, and a dedicated nvidia gpu. I use nvidia-container for cuda containers.

```bash
$ lspci | grep VGA
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
06:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c5)
```
Could this be a problem?

miraculin1, 2024-11-18:
[Solved] Oh, by adding an user that has same UID GID as host system user in Dockerfile, and run rviz as that new user in docker solve this problem.

Docker file:
```Dockerfile
ARG USERNAME=ros
ARG USER_UID=1000
ARG USER_GID=$USER_UID

RUN groupadd --gid $USER_GID $USERNAME \
  && useradd -s /bin/bash --uid $USER_UID --gid $USER_GID -m $USERNAME \
  && mkdir /home/$USERNAME/.config && chown $USER_UID:$USER_GID /home/$USERNAME/.config

RUN echo $USERNAME ALL=\(root\) NOPASSWD:ALL > /etc/sudoers.d/$USERNAME \
  &&chmod 0440 /etc/sudoers.d/$USERNAME
```

tsukione, 2024-11-19:
> [Solved] Oh, by adding an user that has same UID GID as host system user in Dockerfile, and run rviz as that new user in docker solve this problem.
>
> Docker file:
>
> ```dockerfile
> ARG USERNAME=ros
> ARG [...]

saddly, it not working, is there any idea

miraculin1, 2024-11-19:
Hi, have you logged in with the new user, and run rviz as new user?

tsukione, 2024-12-03:
> Hi, have you logged in with the new user, and run rviz as new user?

whether the user name for logging in to the host must be the same as the user name in Docker？

tsukione, 2024-12-03:
https://github.com/ros-visualization/rviz/issues/1628
According to this description, adding `--ipc host` to the `docker run` command solved my problem.