ros-visualization/rviz #1846: Doesn't build with ninja.

cosama, 2025-03-01:
I don't manage to build `rviz` using `ninja` in `catkin`.

```
CMake Error:
  Running

   '/usr/bin/ninja' '-C' '/workspace/build_isolated/rviz' '-t' 'recompact'

  failed with:

   ninja: error: build.ninja:5319: bad $-escape (literal $ must be written as $$)
    COMMAND = cd /workspace/build_isolated/rviz/sip/rviz_sip && $(MAKE)
                                                                ^ near here
```

I am trying to build this from source on `ubuntu:jammy`, so it is a bit of a unusual setup, but it builds fine without `ninja`. To build the workspace I use:

```
catkin_make_isolated --use-ninja -j16 -l16
```

I suspect this is a legit error message and it would really required two $ at that place, but I am not an expert on any o this.

Here some more output:

```
CMake Generate step failed.  Build files cannot be regenerated correctly.
<== Failed to process package 'rviz': 
  Command '['/workspace/devel_isolated/resource_retriever/env.sh', 'cmake', '/workspace/src/rviz', '-DCATKIN_DEVEL_PREFIX=/workspace/devel_isolated/rviz', '-DCMAKE_INSTALL_PREFIX=/workspace/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd /workspace/build_isolated/rviz && /workspace/devel_isolated/resource_retriever/env.sh cmake /workspace/src/rviz -DCATKIN_DEVEL_PREFIX=/workspace/devel_isolated/rviz -DCMAKE_INSTALL_PREFIX=/workspace/install_isolated -G Ninja
```

rhaschke, 2025-03-01:
Looks like the cmake-generated ninja build file is broken. I don't think this is an issue with rviz.

cosama, 2025-03-01:
Yes and no. I think it actually is an issue in `cmake` itself, there is some discussion here: https://gitlab.kitware.com/cmake/cmake/-/issues/18062. But also, I have build tons of ros packages (all of `base` at least) successfully with `ninja` and only this one breaks. It could potentially be circumvented by changing how that particular part is implemented in `cmake`. But I am not an expert in `cmake`, so I just don't know.